Josverl/mpflash #7: mpflash: MacOS ESP-Prog detection hangs

lukaskremla, 2025-02-28:
**Describe the bug**
This bug might not be bound to just MacOS, it might also happen on windows, but I haven't tried it there.

I'm working with an ESP32 S2 chip and I'm communicating with it through JTAG via the [ESP-Prog](https://docs.espressif.com/projects/esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html).

<details>
  <summary>Mpflash fails to identify the ESP32 S2 chip, which I expected, however it hangs on this for a few seconds.</summary>
  

https://github.com/user-attachments/assets/7fd76d52-e4ba-48c5-9c6b-b0bd384550c9

  </details>

Here is the output of:

```Python
import jsons
import serial.tools.list_ports as list_ports

comports = jsons.dumps(list_ports.comports())
print(comports)
```

<details>
  <summary>Formatted json</summary>
  

```json
[
  {
    "description": "n/a",
    "device": "/dev/cu.debug-console",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.debug-console",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "n/a",
    "device": "/dev/cu.Hermes",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.Hermes",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "n/a",
    "device": "/dev/cu.Bluetooth-Incoming-Port",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.Bluetooth-Incoming-Port",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "Espressif Device",
    "device": "/dev/cu.usbmodem2101",
    "hwid": "USB VID:PID=303A:4001 SER=983daee912f0cb3f LOCATION=2-1",
    "interface": null,
    "location": "2-1",
    "manufacturer": "Espressif Systems",
    "name": "cu.usbmodem2101",
    "pid": 16385,
    "product": "Espressif Device",
    "serial_number": "983daee912f0cb3f",
    "vid": 12346
  },
  {
    "description": "Dual RS232-HS",
    "device": "/dev/cu.usbserial-1101",
    "hwid": "USB VID:PID=0403:6010 LOCATION=1-1",
    "interface": null,
    "location": "1-1",
    "manufacturer": "FTDI",
    "name": "cu.usbserial-1101",
    "pid": 24592,
    "product": "Dual RS232-HS",
    "serial_number": null,
    "vid": 1027
  },
  {
    "description": "Dual RS232-HS",
    "device": "/dev/cu.usbserial-1100",
    "hwid": "USB VID:PID=0403:6010 LOCATION=1-1",
    "interface": null,
    "location": "1-1",
    "manufacturer": "FTDI",
    "name": "cu.usbserial-1100",
    "pid": 24592,
    "product": "Dual RS232-HS",
    "serial_number": null,
    "vid": 1027
  }
]
```

  </details>
  
The submodem2101 was identified correctly as an ESP32 S3 and isn't connected via JTAG.
  
Tested through iterm2 and PyCharm with mpflash version 1.24.2. post1 and Python 3.10.11.
 
**Hardware information**
- MacBook M2 Pro
- MacOS Sequoia 15.2
- (This might concern all macOS computers)

lukaskremla, 2025-03-10:
[image]

You should be able to use the 6 pin UART connector of the ESP Prog if you connect it to the corresponding pins on your board.

If you can't see the ESP_EN pin RST should work too, the other pins should be named similarly in the schematics of your board.

Josverl, 2025-03-10:
Ok tried again with a different board : ESP32-PICOKIT_V4 and now I see results  
I use -i to ignore the JTAG port at COM4

<details><summary>mpflash list and flash</summary>
<p>

```pws
PS D:\> mpflash list -i COM4

                            Connected boards
┏━━━━━━┳━━━━━━━━━━━┳━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━┳━━━━━━━┓
┃Serial┃Family     ┃Port ┃Board                          ┃CPU  ┃Version┃
┡━━━━━━╇━━━━━━━━━━━╇━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━╇━━━━━━━┩
│COM5  │micropython│esp32│ESP32_GENERIC                  │ESP32│v1.24.1│
│      │           │     │Generic ESP32 module with ESP32│     │       │
└──────┴───────────┴─────┴───────────────────────────────┴─────┴───────┘
```

Also flashing works fine for this board via the ESP-PROG 

```pws
PS D:\> mpflash flash --serial COM5 --version preview
Error: Failed to get mcu_info for COM4

13:40:02 | ℹ️ Updating ESP32_GENERIC on COM5 to v1.25.0
13:40:02 | ℹ️ Flashing C:\Users\josverl\Downloads\firmware\esp32\ESP32_GENERIC-v1.25.0-preview.365.bin on ESP32_GENERIC on COM5
13:40:02 | ℹ️ Running esptool --chip esp32 --port COM5 erase_flash
esptool.py v4.8.1
Serial port COM5
Connecting....
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None
Crystal is 40MHz
MAC: d8:a0:1d:40:66:5c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.8s
Hard resetting via RTS pin...
13:40:09 | ℹ️ Running esptool --chip esp32 --port COM5 -b 921600 write_flash --compress 0x1000
C:\Users\josverl\Downloads\firmware\esp32\ESP32_GENERIC-v1.25.0-preview.365.bin
esptool.py v4.8.1
Serial port COM5
Connecting...........
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None
Crystal is 40MHz
MAC: d8:a0:1d:40:66:5c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x001a0fff...
Compressed 1701664 bytes to 1116775...
Wrote 1701664 bytes (1116775 compressed) at 0x00001000 in 14.9 seconds (effective 915.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
13:40:30 | ℹ️ Done flashing, resetting the board...
13:40:33 | ✅ Flashed COM5 to ESP32_GENERIC 1.25.0-preview
13:40:33 | ℹ️ Flashed 1 boards
                            Updated boards after flashing
┏━━━━━━┳━━━━━━━━━━━┳━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━┳━━━━━━━━━━━━━━━┳━━━━━┓
┃Serial┃Family     ┃Port ┃Board                          ┃CPU  ┃Version        ┃Build┃
┡━━━━━━╇━━━━━━━━━━━╇━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━╇━━━━━━━━━━━━━━━╇━━━━━┩
│COM5  │micropython│esp32│ESP32_GENERIC                  │ESP32│v1.25.0-preview│  365│
│      │           │     │Generic ESP32 module with ESP32│     │               │     │
└──────┴───────────┴─────┴───────────────────────────────┴─────┴───────────────┴─────┘
PS D:\>
```
</p>
</details> 

One weird thing is that esptool seems to needs to re-neg the protocol 
```
PS D:\> esptool -p COM5 chip_id
esptool.py v4.8.1
Serial port COM5
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision v1.0)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None
Crystal is 40MHz
MAC: d8:a0:1d:40:66:5c
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: d8:a0:1d:40:66:5c
Hard resetting via RTS pin...

```

ESP-PROG --> Board
- ESP_EN  --> EN
- ESP_TXD  -->TX0
- ESP_RXD  --> RX0
- ESP_VDD  --> 3v3
- ESP_GND  --> GND
- ESP_IO0  --> IO0

Josverl, 2025-03-12:
i've  ordered a S2 board to troubleshoot.

Josverl, 2025-03-17:
I now have a [esp32-s2 mini board](https://www.wemos.cc/en/latest/s2/s2_mini.html#features) but an unable to get connection through the esp-prog

IIC on this board the UART0 TXD / RXD are only located inside the USB-C connector ...

 U0TXD 48 I/O/T VDD3P3_CPU U0TXD, GPIO43, CLK_OUT1
 U0RXD 49 I/O/T VDD3P3_CPU U0RXD, GPIO44, CLK_OUT2

what board , and what pins do you use ?
me poking around in the dark with near infinite boards has little chance of success ....

lukaskremla, 2025-03-18:
I'm mainly using the ESP-Prog with custom-made boards that don't expose all pins, just the UART connector for the ESP-Prog. However, the ESP32 S series Devkit C1 style boards should expose all of the necessary pins to test this.

Josverl, 2025-03-18:
In [esp32-s2_datasheet_en.pdf](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf) Pg-14 

# ESP32-S2 
it appears that its CPU Pin 48 & 49 
Name | No. | Type | Powerdomain | Function
-|-|-|-|-
U0TXD |48| I/O/T| VDD3P3_CPU |U0TXD, GPIO43, CLK_OUT1
U0RXD |49| I/O/T| VDD3P3_CPU |U0RXD, GPIO44, CLK_OUT2

are that the same pins as you are using with the ESPProg-UART ?

if you can connect via the esp-prog to flash the esp32-S2 
can you than also use mpremote to that same port to get to the repl ?

# [ESP32-S3](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf)
 - I have one of these 

Name | No. | Type | Powerdomain | Function
-|-|-|-|-
U0TXD | 49 | IO | VDD3P3_CPU | IE,WPU IE,WPU IOMUX
U0RXD| 50 |  IO | VDD3P3_CPU | IE,WPU IE,WPU IOMUX

it appears that its CPU Pin 49 & 50 , but on the board RX / TX is marked between 
41 - 42 - TX - RX - 45 - 46 
likely relates to this: 

UART0
The pins U0TXD and U0RXD that are connected to transmit and receive signals are multiplexed with
GPIO43 ~ GPIO44 via IO MUX, and can also be connected to any GPIO via the GPIO Matrix.

Testing with the ESP32-S3 for now I see a difference with the regular ESP32
When connected with a terminal to --> ESP-Prog --> UART0 
- I do see output 
- but any input sent is ignored / not arriving ate the micropython repl ( terminal seems dead ) 
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x105c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbd8
load:0x403cc700,len:0x2e34
entry 0x403c989c
E (291) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (291) esp_psram: PSRAM enabled but initialization failed. Bailing out.
MicroPython v1.25.0-preview.365.g3823aeb0f on 2025-03-07; Generic ESP32S3 module with ESP32S3
Type "help()" for more information.
>>> 
```
I think I remember something that the boards with native USB/CDC do not 'listen' to UART0 .
that would prevent mpremote/mpflash to communicate with the repl

I also can't get esptool to work with this S3 board via ESP-Prog , with also a msg regarding the TX channel 
```
esptool -p COM36 chip_id
esptool.py v4.8.1
Serial port COM36
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: Download mode successfully detected, but getting no sync reply: The serial TX path seems to be down.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

lukaskremla, 2025-04-19:
Sorry for not replying earlier. I delayed replying until I could sit down to test this and then forgot about it, but I finally got around to it now.

I can confirm that using an ESP-Prog with an ESP board that already has a built-in USB-to-serial converter doesn't really work. I only ever used an ESP-Prog with custom-made hardware that has no built in serial converter and only exposes the pins used by ESP-Prog, and I didn't realize the onboard converter would be conflicting.

On the custom-made hardware the serial port behaves just like any other ESP UART port, I can flash it with esptool, access REPL, it works just fine.

The best way to test it is probably with a bare ESP32 chip and a break out board that doesn't have a UART converter. I have some bare chips around but not the right converter, I can follow up in a few days when I get my hands on one.

closed 2025-06-04 by Josverl