Josverl/mpflash #18: mpflash list (1.24.7) now fails to list any board

scruss, 2025-04-15:
Recently reinstalled via pipx on Python 3.12.7 (Ubuntu):

```
mpflash list
17:30:00 | ⚠️ Traceback (most recent call last):

17:30:00 | ⚠️   File "<frozen runpy>", line 189, in _run_module_as_main

17:30:00 | ⚠️   File "<frozen runpy>", line 148, in _get_module_details

17:30:00 | ⚠️   File "<frozen runpy>", line 112, in _get_module_details

17:30:00 | ⚠️   File 
"/home/scruss/.local/pipx/venvs/mpflash/lib/python3.12/site-packages/mpremote/__
init__.py", line 2, in <module>

17:30:00 | ⚠️     from importlib.metadata import version, PackageNotFoundError

17:30:00 | ⚠️   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 
19, in <module>

17:30:00 | ⚠️     from . import _adapters, _meta

17:30:00 | ⚠️   File "/usr/lib/python3.12/importlib/metadata/_adapters.py", line
5, in <module>

17:30:00 | ⚠️     import email.message

17:30:00 | ⚠️   File "/usr/lib/python3.12/email/message.py", line 15, in <module>

17:30:00 | ⚠️     from email import utils

17:30:00 | ⚠️   File "/usr/lib/python3.12/email/utils.py", line 33, in <module>

17:30:00 | ⚠️     from email._parseaddr import quote

17:30:00 | ⚠️   File "/usr/lib/python3.12/email/_parseaddr.py", line 16, in 
<module>

17:30:00 | ⚠️     import time, calendar

17:30:00 | ⚠️   File "/usr/lib/python3.12/calendar.py", line 94, in <module>

17:30:00 | ⚠️     class _localized_month:

17:30:00 | ⚠️   File "/usr/lib/python3.12/calendar.py", line 96, in 
_localized_month

17:30:00 | ⚠️     _months = 

17:30:00 | ⚠️                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

17:30:00 | ⚠️ AttributeError: 'date' object has no attribute 'strftime'

⠋ Scanning ttyACM0 ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0:00:0117:30:01 | ⚠️ Traceback (most recent call last):

17:30:01 | ⚠️   File "<frozen runpy>", line 189, in _run_module_as_main

17:30:01 | ⚠️   File "<frozen runpy>", line 148, in _get_module_details

17:30:01 | ⚠️   File "<frozen runpy>", line 112, in _get_module_details

17:30:01 | ⚠️   File 
"/home/scruss/.local/pipx/venvs/mpflash/lib/python3.12/site-packages/mpremote/__
init__.py", line 2, in <module>

17:30:01 | ⚠️     from importlib.metadata import version, PackageNotFoundError

17:30:01 | ⚠️   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 
19, in <module>
[... 374 lines cut ...]

17:30:07 | ⚠️     class _localized_month:

17:30:07 | ⚠️   File "/usr/lib/python3.12/calendar.py", line 96, in 
_localized_month

17:30:07 | ⚠️     _months = 

17:30:07 | ⚠️                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

17:30:07 | ⚠️ AttributeError: 'date' object has no attribute 'strftime'

Error: Failed to get mcu_info for /dev/ttyUSB1

      Connected boards      
┏━━━━━━━┳━━━━┳━━━━━┳━━━━━━━┓
┃Ser.   ┃Fam.┃Board┃Version┃
┡━━━━━━━╇━━━━╇━━━━━╇━━━━━━━┩
│ttyACM0│?   │     │       │
│ttyACM1│?   │     │       │
│ttyUSB1│?   │     │       │
└───────┴────┴─────┴───────┘
```

There is a Raspberry Pi Pico 2 W attached to /dev/ttyACM1. mpremote finds it.

Josverl, 2025-04-16:
Likely related to new tags die to the recent release, I'll look into it

scruss, 2025-04-16:
Sorry to put you through a lot. This seems to only affect my rather old (but updated to 24.10, Python 3.12.7 ) Ubuntu box. mpflash works fine on a Raspberry Pi with Python 3.11.2

Josverl, 2025-04-16:
No problem,

I still found and solved another missing dependency issue, so time we'll spent

closed 2025-04-16 by Josverl

scruss, 2025-04-18:
Thanks for fixing this. 

version 1.24.8 does leave an attached CircuitPython device (a [QTPy-Knob](https://github.com/todbot/qtpy-knob) running CircuitPython v9.2.3) in an unusable state, though. I have to disconnect and reconnect the device if I run mpflash

Josverl, 2025-04-19:
The change was related to flashing stm32 on windows, unlikely that this is related to this issue.

Mpflash does try to stop any running scripts though, so perhaps it should reset a device to return it to a running state.