Josverl/mpflash #21: Thanks for fixing this.

Josverl, 2025-04-19:
Thanks for fixing this. 

version 1.24.8 does leave an attached CircuitPython device (a [QTPy-Knob](https://github.com/todbot/qtpy-knob) running CircuitPython v9.2.3) in an unusable state, though. I have to disconnect and reconnect the device if I run mpflash

_Originally posted by @scruss in https://github.com/Josverl/mpflash/issues/18#issuecomment-2815719428_

Josverl, 2025-04-19:
The change was related to flashing stm32 on windows, unlikely that this is related to this issue.

Mpflash does try to stop any running scripts though, so perhaps it should reset a device to return it to a running state.

scruss, 2025-05-29:
mpflash 1.24.8 still leaves this device unusable:

                       Connected boards                    
    ┏━━━━━━━┳━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━┓
    ┃Ser.   ┃Fam.┃Board                           ┃Version┃
    ┡━━━━━━━╇━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━┩
    │ttyACM0│cpy │UNKNOWN_BOARD                   │v9.2.3 │
    │       │    │Adafruit QT Py M0 with samd21e18│       │

scruss, 2025-05-29:
I use pipx, so I can't see it yet

Josverl, 2025-05-30:
Perhaps using `pipx install mpflash --pip-args="--preview"` ?

Josverl, 2025-05-30:
Thanks for sharing the logs
- the crash was during establishing the database, and migration of the existing info on downloaded firmwares. I'll look into the fail to prevent or catch the error.
- for the reset, I assumed that the board could be reset by `mpremote reset`, but that requires it to run MicroPython 😬.
I'll add a special case for CP

Josverl, 2025-06-02:
I tested with Rp2040 running a simple blinky loop, and that output is clean.

Is there anything specific running on your device?

mpremote fails to enter rawrepl, I assume that is on the qty py.
Do you see the same noise with just one board plugged in?

Josverl, 2025-06-02:
no need to remove them , you can have mpflash ignore them using ignore or an environment variable 

- `MPFLASH_IGNORE="/dev/ttyUSB0 /dev/ttyUSB" mpremote list`
- `mpremote list --ignore /dev/ttyUSB0 --ignore /dev/ttyUSB`

scruss, 2025-06-07:
I get the system notification that my computer has mounted the CircuitPython device before the reset is attempted. By this time, the REPL is not in the right state to accept commands.

```
$ mpflash -V list --ignore /dev/ttyUSB0  --ignore /dev/ttyUSB1
15:18:37 | 🐞 version: 1.25.0rc4
15:18:37 | 🐞 filtered_comports: ['/dev/ttyACM0', '/dev/ttyACM1']
15:18:37 | DEBUG /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM0 run 
/home/scruss/.local/share/pipx/venvs/mpflash/lib/python3.13/site-packages/mpflas
h/mpremoteboard/mpy_fw_info.py
15:18:38 | DEBUG /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM0 resume cat :board_info.toml
15:18:38 | DEBUG /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM1 run 
/home/scruss/.local/share/pipx/venvs/mpflash/lib/python3.13/site-packages/mpflas
h/mpremoteboard/mpy_fw_info.py
⠋ Scanning ttyACM1 ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0:00:0115:18:38 | DEBUG /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM1 resume cat :board_info.toml

                   Connected boards                    
┏━━━━━━━┳━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━┓
┃Ser.   ┃Fam.┃Board                           ┃Version┃
┡━━━━━━━╇━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━┩
│ttyACM0│mpy │RPI_PICO                        │v1.25.0│
│       │    │Raspberry Pi Pico with RP2040   │       │
│ttyACM1│cpy │UNKNOWN_BOARD                   │v9.2.3 │
│       │    │Adafruit QT Py M0 with samd21e18│       │
└───────┴────┴────────────────────────────────┴───────┘
15:18:39 | 🐞 /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM0 resume reset
15:18:39 | 🐞 /home/scruss/.local/share/pipx/venvs/mpflash/bin/python -m 
mpremote connect /dev/ttyACM1 exec --no-follow import 
microcontroller,time;time.sleep(0.01);microcontroller.reset()
15:18:39 | ⚠️ mpremote: failed to access /dev/ttyACM1 (it may be in use by 
another program)

```